illiliti/tinyramfs #4: Cant stat '/lib//usr/lib/libc.so'

muevoid, 2020-04-18:
Hello! This is my first time messing around with the initramfs on my personal machine. When running it with lvm and luks enabled I get that error. I also get the following two errors.

cp: can't create '/tmp/initramfs.31940/lib/modules/5.4.33-gnu/modules.builtin': Path does not exist

And

cp: can't create '/tmp/initramfs.31940/lib/modules/5.4.33-gnu/modules.dep': Path does not exist

illiliti, 2020-04-18:
Hmm, strange. Can you enable debug mode (--debug) and post output here?

muevoid, 2020-04-18:
Here is the error link: https://termbin.com/hpg1

Or plain text:

+ trap '
        ret="$?"
        trap - EXIT INT
        [ "$debug" != 1 ] && remove_workdir
        [ "$ret"   != 0 ] && [ "$panic" != 1 ] && panic "something went wrong"
        ' EXIT INT
+ create_structure
+ info 'creating directory structure'
+ printf 'info >> %s\n' 'creating directory structure'
+ mkdir -p /tmp/initramfs.29303/etc /tmp/initramfs.29303/dev /tmp/initramfs.29303/sys /tmp/initramfs.29303/tmp /tmp/initramfs.29303/proc /tmp/initramfs.29303/root /tmp/initramfs.29303/usr/lib /tmp/initramfs.29303/usr/bin /tmp/initramfs.29303/mnt/root
+ create_symlinks
+ info 'creating symlinks'
+ printf 'info >> %s\n' 'creating symlinks'
+ ln -s usr/lib /tmp/initramfs.29303/lib
+ ln -s usr/lib /tmp/initramfs.29303/lib64
+ ln -s usr/bin /tmp/initramfs.29303/bin
+ ln -s usr/bin /tmp/initramfs.29303/sbin
+ ln -s bin /tmp/initramfs.29303/usr/sbin
+ ln -s lib /tmp/initramfs.29303/usr/lib64
+ '[' 1 '=' 1 ]
+ install_lvm
+ info 'installing LVM'
+ printf 'info >> %s\n' 'installing LVM'
+ install_binary lvchange
+ command -v lvchange
+ binary=/usr/bin/lvchange
+ : no operation
+ '[' -e /tmp/initramfs.29303/usr/bin/lvchange ]
+ '[' -h /usr/bin/lvchange ]
+ cp -P /usr/bin/lvchange /tmp/initramfs.29303/usr/bin/
+ readlink /usr/bin/lvchange
+ binary=/usr/bin/lvm
+ '[' -h /usr/bin/lvm ]
+ install -m755 /usr/bin/lvm /tmp/initramfs.29303/usr/bin/lvm
+ strip /tmp/initramfs.29303/usr/bin/lvm
+ ldd /usr/bin/lvm
+ read -r _library
+ ldd /usr/bin/lvm
+ _library='/lib/ld-musl-x86_64.so.1 (0x7f0a5d686000)'
+ _library='/lib/ld-musl-x86_64.so.1 (0x7f0a5d686000)'
+ _library=/lib/ld-musl-x86_64.so.1
+ '[' -e /lib/ld-musl-x86_64.so.1 ]
+ install_library /lib/ld-musl-x86_64.so.1
+ library=/lib/ld-musl-x86_64.so.1
+ '[' -e /tmp/initramfs.29303/usr/lib/ld-musl-x86_64.so.1 ]
+ '[' -h /lib/ld-musl-x86_64.so.1 ]
+ cp -P /lib/ld-musl-x86_64.so.1 /tmp/initramfs.29303/usr/lib/
+ readlink /lib/ld-musl-x86_64.so.1
+ library=/lib//usr/lib/libc.so
+ '[' -h /lib//usr/lib/libc.so ]
+ install -m755 /lib//usr/lib/libc.so /tmp/initramfs.29303/usr/lib/libc.so
install: can't stat '/lib//usr/lib/libc.so': No such file or directory
+ strip /tmp/initramfs.29303/usr/lib/libc.so
+ :
+ read -r _library
+ _library='libdevmapper-event.so.1.02 => /lib/libdevmapper-event.so.1.02 (0x7f0a5d217000)'
+ _library='/lib/libdevmapper-event.so.1.02 (0x7f0a5d217000)'
+ _library=/lib/libdevmapper-event.so.1.02
+ '[' -e /lib/libdevmapper-event.so.1.02 ]
+ install_library /lib/libdevmapper-event.so.1.02
+ library=/lib/libdevmapper-event.so.1.02
+ '[' -e /tmp/initramfs.29303/usr/lib/libdevmapper-event.so.1.02 ]
+ '[' -h /lib/libdevmapper-event.so.1.02 ]
+ install -m755 /lib/libdevmapper-event.so.1.02 /tmp/initramfs.29303/usr/lib/libdevmapper-event.so.1.02
+ strip /tmp/initramfs.29303/usr/lib/libdevmapper-event.so.1.02
+ read -r _library
[... 449 lines cut ...]
+ cp -P /usr/bin/switch_root /tmp/initramfs.29303/usr/bin/
+ readlink /usr/bin/switch_root
+ binary=/usr/bin/busybox
+ '[' -h /usr/bin/busybox ]
+ install -m755 /usr/bin/busybox /tmp/initramfs.29303/usr/bin/busybox
+ strip /tmp/initramfs.29303/usr/bin/busybox
+ ldd /usr/bin/busybox
+ return 0
+ install -m755 /usr/share/tinyramfs/init /tmp/initramfs.29303/init
+ printf '%s\n' 'root=' 'root_type=' 'root_opts=' 'devmgr=' 'monolith=1'
+ create_initramfs
+ info 'creating initramfs image'
+ printf 'info >> %s\n' 'creating initramfs image'
+ '[' 0 '!=' 1 ]
+ '[' -e /boot/tinyramfs-5.4.33.img ]
+ panic 'initramfs image already exist'
+ printf 'panic >> %s\n' 'initramfs image already exist'
panic >> initramfs image already exist
+ panic=1
+ exit 1
+ ret=1
+ trap - EXIT INT
+ '[' 1 '!=' 1 ]
+ '[' 1 '!=' 0 ]
+ '[' 1 '!=' 1 ]

muevoid, 2020-04-18:
I just noticed it still created an initramfs despite that error. I will attempt booting with it.

After doing that I get another error from the init file saying line 67 ln: not found then ends in kernel panic.

illiliti, 2020-04-18:
Fixed

muevoid, 2020-04-18:
Kk testing right now

muevoid, 2020-04-18:
Closer however now I recieve an error saying panic >> failed to lookup parition after typing in my passphrase it hangs for roughly 10 seconds before this. Then states sh: cant access tty; job control turned off. FYI I do have a second grub menu entry with better initramfs currently so I do have a working pc it would be nice to switch to this though.

illiliti, 2020-04-18:
Which device manager do you use? (mdev,mdev,udev) Looks like decryption process is very slow, so that caused 'failed to lookup partition'. I can increase delay if that's root of problem

muevoid, 2020-04-18:
I use eudev currently

illiliti, 2020-04-18:
I have increased delay to 80

muevoid, 2020-04-18:
Same error unfortunately

illiliti, 2020-04-18:
Are you sure that your config correct?

muevoid, 2020-04-18:
Im pretty sure. Here it is:
#
# configuration
#

# debug mode
#
debug=0

# overwrite initramfs
#
force=0

# initramfs output path
#
# default - /tmp/initramfs-$kernel
# example - output="/tmp/myinitramfs.img.gz"
#
output="/boot/tinyramfs-5.4.33.img"

# monolithic kernel
#
monolith=1

# modules directory
#
# default - /lib/modules
# example - moddir="/mnt/root/lib/modules"
#
moddir=""

# kernel version
#
# default - $(uname -r)
# example - kernel="5.4.18_1"
#
kernel=""

# compression program
#
# default - gzip -9
# example - compress="pigz -9"
#
compress=""

# root
#
# supported - PARTUUID, DEVICE, LABEL, UUID
# example -
#   root="/dev/sda1"
#   root="PARTUUID=35f923c5-083a-4950-a4da-e611d0778121"
#
root="/dev/mapper/vg-root"

# root type
#
# default - autodetected
# example - root_type="btrfs"
#
root_type=""

# root options
# example - see fstab(5)
#
root_opts=""

# device manager
# supported - udev, mdev, mdevd
#
devmgr="udev"

# hostonly mode
#
hostonly=0

# additional modules
# example - modules="fat crc32c_generic"
#
modules=""

# exclude modules
# example - modules_exclude="wmi fuse"
#
modules_exclude=""

# additional binaries
# example - binaries="ls cat /path/to/mycustomprog"
#
binaries=""

# LVM support
#
lvm=1

# LVM options
#
# supported - tag, name, group, config, discard
# description -
#   tag - trigger lvm by tag
#   name - trigger lvm by logical volume name
#   group - trigger lvm by volume group name
#   config - embed host lvm config
#   discard - enable issue_discards
# example -
#   lvm_opts="tag=lvm-server"
#   lvm_opts="name=lv1,group=vg1"
#   lvm_opts="config=1,discard"
#   lvm_opts="discard=1"
#
lvm_opts="discard=1"

# LUKS support
#
luks=1

# LUKS encrypted root
#
# supported - PARTUUID, DEVICE, LABEL, UUID
# example -
#   luks_root="/dev/sda1"
#   luks_root="PARTUUID=35f923c5-083a-4950-a4da-e611d0778121"
#
luks_root="/dev/sda2"

# LUKS options
#
# supported - key, name, header, discard
# description -
#   key - embed key
#   name - device mapper name
#   header - embed header
#   discard - enable allow-discards
# example -
#luks_opts="key=/path/to/keyfile,name=myluksroot,header=/path/to/header,discard"
#   luks_opts="discard=1"
#
luks_opts="discard=1,name=luks"

illiliti, 2020-04-18:
Try to use UUID or PARTUUID instead of /dev/

muevoid, 2020-04-18:
Kk ill try that out tommorow unfortunately dont have any more time to debug currently thanks for all the help so far!

illiliti, 2020-04-18:
Fine, glad to help either way. I'll leave it open until your issues being fixed

illiliti, 2020-04-18:
I think i figured out what caused it. Fix pushed

muevoid, 2020-04-18:
Yep that worked! Thank you so much for the help Im going to do more testing with some different configurations to make sure everything still works out but this is very promising!

closed 2020-04-18 by muevoid