illiliti/tinyramfs #6: Hanging

muevoid, 2020-05-17:
Unfortunately after typing my passphrase it appears to hang. Any help would be appeciated. My system is fully updated. I only use luks no lvm.

illiliti, 2020-05-17:
Which luks version do you use(LUKS1 or LUKS2)? Can you try to boot with kernel parameter `break=luks` and manually unlock luks with `--debug`?

illiliti, 2020-05-17:
Also, your processor is AMD Ryzen?

muevoid, 2020-05-17:
I will attempt that in a little bit in pc I can answer the other two questions though. Pretty sure luk2 and no ive had it working on this pc before last time I installed kiss not sure what changed.

illiliti, 2020-05-17:
Json-c update may be related to this issue
https://bbs.archlinux.org/viewtopic.php?id=255283
https://gitlab.com/cryptsetup/cryptsetup/-/issues/558
https://github.com/json-c/json-c/issues/588

muevoid, 2020-05-17:
Ah I think that may be it I do have discards enabled I will try re making tinyramfs with no discards

illiliti, 2020-05-26:
Well..Did you resolve this issue?

illiliti, 2020-06-28:
Fixed

closed 2020-06-28 by illiliti