illiliti/tinyramfs #9: Boot Issue: Tinyramfs cannot find root partition after init initramfs step.

nightgaunt, 2020-07-05:
Hey there! 
I'm having some trouble getting my machine to boot. I've gotten to the point where grub tries to load the initramfs and then it fails to find the root partition.  I'm executing tinyramfs with: 
tinyramfs -k 5.7.7 -o /boot/initramfs-linux.img (with adding -f if I'm overwriting).

I've exhausted my problem solving capabilities on this one and would love to see if someone has an idea of where I'm going wrong. :)

blkid -f
```
sdb
├─sdb1                    vfat        FAT16          8DE7-EBA9                               499.5M     0% /mnt/boot/efi
├─sdb2                    crypto_LUKS 1              ca8ebf4f-6e11-4fa7-93ad-af96ca49ccd7
│ └─boot                  ext2        1.0            6290cc6c-0b1a-4e83-89e8-ccad9ea304bc      1.9G     1% /mnt/boot
└─sdb3                    crypto_LUKS 2              b0008a0c-8cc3-42d7-aab2-ca2acdc1e956
  └─root                  ext4        1.0            de28ef56-90ba-4a91-a358-f5f687ada697    202.5G     1% /mnt

```

Here is the config for Tinyramfs:

```
# default - 0
# enable this if you have monolithic kernel (builtin modules)
#
monolith=0

# compression program
#
# supported - whatever you choose
# default - none
# example - compress="pigz -9"
#
compress="none"

# root file system
#
# supported - UUID, LABEL, DEVICE, PARTUUID
# example -
#   root="/dev/sda1"
#   root="/dev/dm-0"
#   root="/dev/disk/by-uuid/13bcb7cc-8fe5-4f8e-a1fe-e4b5b336f3ef"
#   root="PARTUUID=35f923c5-083a-4950-a4da-e611d0778121"
#
root="UUID=b0008a0c-8cc3-42d7-aab2-ca2acdc1e956"

# root file system type
#
# default - autodetected throught /proc/mounts
# example - root_type="btrfs"
#
root_type="ext4"

# root file system options
# example - root_opts="subvol=mysubvolume,nodatacow,defaults"
# see fstab(5) man page for more info
#
root_opts="defaults"

# device manager
# supported - udev, mdev, mdevd
#
devmgr="udev"

# hostonly mode
#
# default - 0
# dramatically reduce initramfs size
# useful only for modular kernels
#
hostonly=0

# hooks
#
# currently supported - lvm, luks
# example -
#   hooks="lvm luks" will support booting LUKS on LVM
#   hooks="luks lvm" will support booting LVM on LUKS
#   hooks="luks"     will support booting only LUKS
#   and so on...
#
hooks="luks"

# LVM options
#
# supported - tag, name, group, config, discard
# description -
#   tag - trigger lvm by tag
#   name - trigger lvm by logical volume name. group must be specified
#   group - trigger lvm by volume group name
#   config - embed host lvm config
#   discard - enable issue_discards
# example -
#   lvm_opts="tag=lvm-server"
#   lvm_opts="name=lv1,group=vg1"
#   lvm_opts="config=1,discard=1"
#   lvm_opts="discard=1"
#
#lvm_opts=""

# LUKS options
#
# supported - key, name, root, header, discard
# description -
#   key - embed key
#   name - device mapper name
#   root - encrypted root ( UUID, LABEL, DEVICE, PARTUUID )
#   header - embed header
#   discard - enable allow-discards
# example -
#   luks_opts="root=PARTUUID=35f923c5-083a-4950-a4da-e611d0778121,key=/path/to/keyfile,name=myluksroot,header=/path/to/header,discard=1"
#   luks_opts="root=PARTUUID=35f923c5-083a-4950-a4da-e611d0778121,
#              key=/path/to/keyfile,
#              name=myluksroot,
#              header=/path/to/header,
#              discard=1"
#
#   luks_opts="root=/dev/sda1,discard=1"
#
luks_opts="root=UUID=b0008a0c-8cc3-42d7-aab2-ca2acdc1e956,key=/boot/rootkeyfile.key,name=root,discard=1"
```

illiliti, 2020-07-06:
You have a typo in `root`. UUID points to luks container, but should point to real root. Change root to `root=UUID=de28ef56-90ba-4a91-a358-f5f687ada697`

`none` in `compress` is not supported option. leave it commented or use gzip(recommended)
`root_opts="defaults"` no need

And, i assume you using busybox modprobe ? If so you need to compile `ext4` built in (CONFIG_EXT4_FS=y), or manually include necessary modules using hooks.
Example:
1. Add to /etc/tinyramfs/hooks/ext4/ext4
```sh
for _mod in crc32c libcrc32c; do
    copy_module "$_mod"
done
```
2. Add to /etc/tinyramfs/hooks/ext4/ext4.init.early
```
modprobe -a crc32c libcrc32c
```
3. Append `ext4` to `hooks` option

This needed because busybox modprobe doesn't handle soft dependencies unlike kmod modprobe(you can install `kmod` to get rid of this manual intervention btw)

_Everything is confusing because documentation is lacking i know, i'm working on it_

nightgaunt, 2020-07-08:
Progress report! It's been kinda crazy lately so I haven't been at my computer. Unfortunately that fix didn't work. I installed
d KMOD to bypass the manual intervention, however that didn't work either so I did the manual intervention that you mentioned (added the hook, created the files ext4 and ext4.init.early. Still stuck at a failure to lookup partition. I feel like I'm being a goofball and missing something or not typing something in right. Hopefully that's the case.

Thank you for your time!

`
sdb
├─sdb1                    vfat                    8DE7-EBA9                               499.5M     0% /boot/efi
├─sdb2                    crypto_LUKS             ca8ebf4f-6e11-4fa7-93ad-af96ca49ccd7
│ └─boot                  ext2                    6290cc6c-0b1a-4e83-89e8-ccad9ea304bc      1.9G     1% /boot
└─sdb3                    crypto_LUKS             b0008a0c-8cc3-42d7-aab2-ca2acdc1e956
  └─root                  ext4                    de28ef56-90ba-4a91-a358-f5f687ada697    202.5G     1% /
`

Okay this is what my new config file looks like.

`
#
# configuration
#
# uncomment and fill settings which you needed
#
# 0 - disable
# 1 - enable

# monolithic kernel
#
# default - 0
# enable this if you have monolithic kernel (builtin modules)
#
monolith=0

# compression program
#
# supported - whatever you choose
# default - none
# example - compress="pigz -9"
#
compress="gzip"

# root file system
#
# supported - UUID, LABEL, DEVICE, PARTUUID
# example -
#   root="/dev/sda1"
#   root="/dev/dm-0"
#   root="/dev/disk/by-uuid/13bcb7cc-8fe5-4f8e-a1fe-e4b5b336f3ef"
#   root="PARTUUID=35f923c5-083a-4950-a4da-e611d0778121"
#
root="UUID=de28ef56-90ba-4a91-a358-f5f687ada697"

# root file system type
#
# default - autodetected throught /proc/mounts
# example - root_type="btrfs"
#
root_type="ext4"

# root file system options
# example - root_opts="subvol=mysubvolume,nodatacow,defaults"
# see fstab(5) man page for more info
#
#root_opts="defaults"

# device manager
# supported - udev, mdev, mdevd
#
devmgr="udev"

# hostonly mode
#
# default - 0
# dramatically reduce initramfs size
# useful only for modular kernels
#
hostonly=0

# hooks
#
# currently supported - lvm, luks
# example -
#   hooks="lvm luks" will support booting LUKS on LVM
#   hooks="luks lvm" will support booting LVM on LUKS
#   hooks="luks"     will support booting only LUKS
#   and so on...
#
hooks="luks,ext4"

# LVM options
#
# supported - tag, name, group, config, discard
# description -
#   tag - trigger lvm by tag
#   name - trigger lvm by logical volume name. group must be specified
#   group - trigger lvm by volume group name
#   config - embed host lvm config
#   discard - enable issue_discards
# example -
#   lvm_opts="tag=lvm-server"
#   lvm_opts="name=lv1,group=vg1"
#   lvm_opts="config=1,discard=1"
#   lvm_opts="discard=1"
#
#lvm_opts=""

# LUKS options
#
# supported - key, name, root, header, discard
# description -
#   key - embed key
#   name - device mapper name
#   root - encrypted root ( UUID, LABEL, DEVICE, PARTUUID )
#   header - embed header
#   discard - enable allow-discards
# example -
#   luks_opts="root=PARTUUID=35f923c5-083a-4950-a4da-e611d0778121,key=/path/to/keyfile,name=myluksroot,header=/path/to/header,
discard=1"
`
Here is my grub.cfg:
`
### BEGIN /etc/grub.d/10_linux ###
menuentry 'KISS GNU/Linux' --class kiss --class gnu-linux --class gnu --class os $menuentry_id_option 'gnulinux-simple-de28ef5
6-90ba-4a91-a358-f5f687ada697' {
        load_video
        insmod gzio
        insmod part_msdos
        insmod cryptodisk
        insmod luks
        insmod gcry_rijndael
        insmod gcry_rijndael
        insmod gcry_sha256
        insmod ext2
        cryptomount -u ca8ebf4f6e114fa793adaf96ca49ccd7
        set root='cryptouuid/ca8ebf4f6e114fa793adaf96ca49ccd7'
        if [ x$feature_platform_search_hint = xy ]; then
          search --no-floppy --fs-uuid --set=root --hint='cryptouuid/ca8ebf4f6e114fa793adaf96ca49ccd7'  6290cc6c-0b1a-4e83-89e
8-ccad9ea304bc
        else
          search --no-floppy --fs-uuid --set=root 6290cc6c-0b1a-4e83-89e8-ccad9ea304bc
        fi
        echo    'Loading Linux 5.7.7 ...'
        linux   /vmlinuz-5.7.7 root=UUID=de28ef56-90ba-4a91-a358-f5f687ada697 ro  loglevel=3 quiet
        echo    'Loading initial ramdisk ...'
        initrd  /initramfs-5.7.7.img
}
`

illiliti, 2020-07-09:
> hooks="luks,ext4"

space separated, not comma

> I installed
> d KMOD to bypass the manual intervention, however that didn't work either

After installing kmod you need to replace busybox modprobe and depmod with kmod variants and run `depmod` command

If that didn't help you need to boot with `break=root` kernel parameter and manually try to mount root fs, e.g `busybox mount -t ext4 /dev/disk/by-uuid/de28ef56-90ba-4a91-a358-f5f687ada697 /mnt/root`

If file /dev/disk/by-uuid/de28ef56-90ba-4a91-a358-f5f687ada697 didn't exist run `blkid` or `busybox blkid` to see which partitions currently available. If blkid didn't find your root fs, can you send me your kernel config ? `cat /proc/config.gz | gzip -d`

nightgaunt, 2020-07-10:
I'm having trouble replacing modprobe and depmod with the kmod variants because the kiss-chroot pulls uname -r which in my case is my host machine and not the kernel that kiss is using. Modprobe and depmod both throw errors that they can't find the kernel folder and I'm not sure if they are working properly. Did you have any trouble with this?

nightgaunt, 2020-07-10:
kernel config
https://termbin.com/mn6o

nightgaunt, 2020-07-11:
hey illiliti, looks like after updating the keyfile on my boot partition, tinyramfs can't find the partition again. What was that command that you gave me on IRC for initializing the program from the git repo itself starting with ./tinyramfs?

illiliti, 2020-07-11:
https://freenode.logbot.info/kisslinux/20200710#c4353515

nightgaunt, 2020-07-11:
Ah, yeah. I'm at a loss then. It gave me a "Unable to find key file" yesterday and when I recreated the key (cryptsetup luksAddKey.../dev/disk) it just reverted back to throwing the failed to lookup partition.

closed 2020-09-15 by crushingteeth