illiliti/tinyramfs #12: stuck at "copying all modules"

TheBrainScrambler, 2020-08-29:
Hello again,
I am currently running this in a KISS Linux chroot with busybox tools, cryptsetup, kmod etc ...
When I run `sudo tinyramfs -o /boot/initramfs -k 5.8.3` (-k is needed because the chroot doesn't have the same kernel as the host) I'm getting this log:
```
>> preparing environment
>> preparing initramfs
>> running keymap hook
>> configuring keymap
>> running eudev hook
>> configuring eudev
>> running luks hook
>> configuring LUKS
>> copying all modules
cp: can't create '/tmp/tinyramfs.11288/lib/modules/5.8.3/modules.builtin': Path does not exist
cp: can't create '/tmp/tinyramfs.11288/lib/modules/5.8.3/modules.order': Path does not exist
```

illiliti, 2020-08-29:
run tinyramfs with `--debug` and send me full log
also send me output of `ls <chroot_dir>/lib/modules/5.8.3/*`

TheBrainScrambler, 2020-08-30:
`tinyramfs -k 5.8.3 --debug > tinyramfs.log 2>&1`
http://0x0.st/iE9v.log

`ls /lib/modules/5.8.3/* > modules.log`
http://0x0.st/iE9w.log
(yes this is the folder of the chroot, not the host)

illiliti, 2020-08-30:
looks like you have monolithic kernel(add 'monolith=1' to tinyramfs config) or you forgot to install modules(make INSTALL_MOD_STRIP=1 modules_install)

TheBrainScrambler, 2020-08-31:
I don't really understand my problem: I didn't had a monolithic kernel, but modules. But when I do `make modules_install` there's no modules that get installed ... And then I'm trying to get them built-in in the kernel, but that doesn't seem to work either ... I'm closing the issue, I don't think tinyramfs is responsible for all this.

closed 2020-08-31 by TheBrainScrambler