illiliti/tinyramfs #13: Another infinite loop

quisys, 2020-08-29:
Debug log:

```
033[1;37m>>033[m preparing environment
+ prepare_initramfs
+ print 'preparing initramfs'
+ printf '%b %s\n' '033[1;37m>>033[m' 'preparing initramfs'
033[1;37m>>033[m preparing initramfs
+ mkdir -p /tmp/tinyramfs.7204/dev /tmp/tinyramfs.7204/sys /tmp/tinyramfs.7204/tmp /tmp/tinyramfs.7204/run /tmp/tinyramfs.7204/var /tmp/tinyramfs.7204/proc /tmp/tinyramfs.7204/root /tmp/tinyramfs.7204/usr/lib /tmp/tinyramfs.7204/usr/bin /tmp/tinyramfs.7204/mnt/root /tmp/tinyramfs.7204/etc/tinyramfs
+ ln -s usr/lib /tmp/tinyramfs.7204/lib
+ ln -s usr/bin /tmp/tinyramfs.7204/bin
+ ln -s usr/bin /tmp/tinyramfs.7204/sbin
+ ln -s ../run /tmp/tinyramfs.7204/var/run
+ ln -s ../run/lock /tmp/tinyramfs.7204/var/lock
+ ln -s bin /tmp/tinyramfs.7204/usr/sbin
+ for _binary in \[ sh ln env kill mkdir sleep mount printf switch_root "${srcdir}/device-helper"
+ copy_binary '['
++ command -v '['
+ binary='['
+ case "$binary" in
+ IFS=:
+ set -- /usr/local/sbin /usr/local/bin /usr/sbin /usr/bin /sbin /bin /opt/bin /usr/lib/llvm/10/bin
+ unset IFS
+ for _dir in "$@"
+ '[' -x '/usr/local/sbin/[' ']'
+ continue
+ for _dir in "$@"
+ '[' -x '/usr/local/bin/[' ']'
+ continue
+ for _dir in "$@"
+ '[' -x '/usr/sbin/[' ']'
+ continue
+ for _dir in "$@"
+ '[' -x '/usr/bin/[' ']'
+ binary='/usr/bin/['
+ break
+ '[' -e '/tmp/tinyramfs.7204/bin/[' ']'
+ '[' -h '/usr/bin/[' ']'
+ cp '/usr/bin/[' /tmp/tinyramfs.7204/bin
+ chmod 755 '/tmp/tinyramfs.7204/bin/['
+ strip '/tmp/tinyramfs.7204/bin/['
+ ldd '/usr/bin/['
+ read -r _library
+ _library='linux-vdso.so.1 (0x00007ffddb75d000)'
+ _library=linux-vdso.so.1
+ '[' -e linux-vdso.so.1 ']'
+ continue
+ read -r _library
+ _library='/lib64/libc.so.6 (0x00007f12b4c5d000)'
+ _library=/lib64/libc.so.6
+ '[' -e /lib64/libc.so.6 ']'
+ copy_library /lib64/libc.so.6
+ library=/lib64/libc.so.6
+ '[' -e /tmp/tinyramfs.7204/lib/libc.so.6 ']'
+ '[' -h /lib64/libc.so.6 ']'
++ ls -ld /lib64/libc.so.6
+ symlink='lrwxrwxrwx 1 root root 12 Aug 28 20:51 /lib64/libc.so.6 -> libc-2.32.so'
+ cp -P /lib64/libc.so.6 /tmp/tinyramfs.7204/lib
+ library=/lib/libc-2.32.so
+ '[' -h /lib/libc-2.32.so ']'
+ cp /lib/libc-2.32.so /tmp/tinyramfs.7204/lib
+ chmod 755 /tmp/tinyramfs.7204/lib/libc-2.32.so
+ strip /tmp/tinyramfs.7204/lib/libc-2.32.so
+ read -r _library
+ _library='/lib64/ld-linux-x86-64.so.2 (0x00007f12b4e39000)'
+ _library=/lib64/ld-linux-x86-64.so.2
+ '[' -e /lib64/ld-linux-x86-64.so.2 ']'
+ copy_library /lib64/ld-linux-x86-64.so.2
+ library=/lib64/ld-linux-x86-64.so.2
+ '[' -e /tmp/tinyramfs.7204/lib/ld-linux-x86-64.so.2 ']'
[... 301 lines cut ...]
+ cp -P /usr/bin/env /tmp/tinyramfs.7204/bin
+ binary=/usr/bin/env
+ '[' -h /usr/bin/env ']'
++ ls -ld /usr/bin/env
+ symlink='lrwxrwxrwx 1 root root 13 Aug 16 18:20 /usr/bin/env -> ../../bin/env'
+ cp -P /usr/bin/env /tmp/tinyramfs.7204/bin
+ binary=/usr/bin/env
+ '[' -h /usr/bin/env ']'
++ ls -ld /usr/bin/env
+ symlink='lrwxrwxrwx 1 root root 13 Aug 16 18:20 /usr/bin/env -> ../../bin/env'
+ cp -P /usr/bin/env /tmp/tinyramfs.7204/bin
+ binary=/usr/bin/env
+ '[' -h /usr/bin/env ']'
++ ls -ld /usr/bin/env
+ symlink='lrwxrwxrwx 1 root root 13 Aug 16 18:20 /usr/bin/env -> ../../bin/env'
+ cp -P /usr/bin/env /tmp/tinyramfs.7204/bin
+ binary=/usr/bin/env
+ '[' -h /usr/bin/env ']'
++ ls -ld /usr/bin/env
+ symlink='lrwxrwxrwx 1 root root 13 Aug 16 18:20 /usr/bin/env -> ../../bin/env'
+ cp -P /usr/bin/env /tmp/tinyramfs.7204/bin
+ binary=/usr/bin/env
+ '[' -h /usr/bin/env ']'
++ ls -ld /usr/bin/env
```

illiliti, 2020-08-29:
Same problem. While writing this function i didn't realize that symlink can be relative, my bad

quisys, 2020-08-29:
Thanks for the fix

illiliti, 2020-08-29:
I didn't submit fix, will do later

illiliti, 2020-08-29:
should be fixed. can you confirm ?

quisys, 2020-09-01:
yes

closed 2020-09-01 by quisys