illiliti/tinyramfs #16: Add overlayfs support

eudaldgr, 2020-09-10:
I'm unable to boot, due to `tinyramfs` don't mount overlayfs on init as it needs.

eudaldgr, 2020-09-10:
I made a simple [sed](https://github.com/eudaldgr/kiss-live/blob/master/kiss-live#L160) modification to work, but it's only a dirty workaround.
Here, you can check my `tinyramfs` [config](https://github.com/eudaldgr/kiss-live/blob/master/files/config), and [hooks](https://github.com/eudaldgr/kiss-live/tree/master/hooks).
Note, that with this fs i don't need to point to a 'root=...' on the config, but without it won't boot.

illiliti, 2020-09-10:
fixed. now you can specify `root` option to pseudofs. for example: `root=overlay`

closed 2020-09-10 by illiliti

eudaldgr, 2020-09-11:
I did not explain very well, but you have understood it perfectly. :)

commit 4f94d6e4 (2020-09-10): allow setting root option to pseudofs. fixes #16