illiliti/tinyramfs #21: Can't boot LVM

Ominitay, 2020-12-02:
Hi there! I'm trying to use this to get Kiss Linux working with an LVM root partition. However, when attempting to boot, the kernel panics, as the `root=UUID=...` kernel parameter is set to the UUID of the LVM partition (which I think it should be), but the kernel can't seem to find the LV. My kernel parameters (in `refind_linux.conf`) are: `rw root=UUID=[LVM UUID] initrd=EFI\kiss\initramfs-linux59`. A screenshot of the kernel panic (from my VM) is attached. Also provided is my `/etc/tinyramfs/config` file.
[image]
```
compress=xz

root=UUID=2b24bf94-d23c-4dd5-9a61-5501c3f1accc
root_type=ext4

hooks="eudev lvm"

lvm_config=1
lvm_discard=1
lvm_name=root
lvm_group=KissLV
```
I'm probably missing something obvious, but I've never used this tool before. Thank you for any help in advance!

eudaldgr, 2020-12-02:
I have a similar issue when I compress the initrd without 'check=crc32', try `compress="xz --check=crc32"`

Ominitay, 2020-12-02:
> I have a similar issue when I compress the initrd without 'check=crc32', try `compress="xz --check=crc32"`

Thank you for the idea! I will be able to try this in a few minutes...

illiliti, 2020-12-02:
This is intended behavior because tinyramfs doesn't handle the quirks of the compression program, but only provides the ability to use it via config.

closed 2020-12-02 by illiliti