illiliti/tinyramfs #22: Mount fails with device not found error.

Ominitay, 2021-02-16:
I am trying to run Kiss from a BTRFS partition on an LVM logical volume. I get dropped into an emergency shell, with the error that the device being mounted doesn't exist. I can see the device in `/dev`. From the emergency shell, I get the same error if I try to mount the root partition manually, though this doesn't happen to the non-lvm, non-btrfs, EFI partition (that mounts fine).

My config is:
```
compress="xz --check=crc32"

root="UUID=042f9fb5-6657-47d9-b81a-cebd86e9b94f"
root_type=btrfs

hooks="eudev lvm"

lvm_config=1
lvm_name=kissroot
lvm_group=kissvg
```

The output of `blkid | grep kissroot` is 
```
/dev/mapper/kissvg-kissroot: UUID="042f9fb5-6657-47d9-b81a-cebd86e9b94f" UUID_SUB="895e196c-be51-44f5-ad78-7bca26325cec" BLOCK_SIZE="4096" TYPE="btrfs"
```

Ominitay, 2021-02-16:
[image]
This is the output I get ^

illiliti, 2021-02-16:
I'm pretty sure something wrong with kernel configuration or partition layout. Look, `mount(2)` fails with ENODEV error (No such device). According to https://man7.org/linux/man-pages/man2/mount.2.html this means that filesystem is not supported in kernel or corresponding module wasn't loaded.

Are you using kernel with builtin or external modules?

Ominitay, 2021-02-16:
Oh, I have modules in /lib/modules/5.10.16, so I'll try rebuilding with monolithic=0...

Ominitay, 2021-02-16:
Unfortunately that did not fix it. Do you think I may have misconfigured my kernel perhaps?

Ominitay, 2021-02-16:
:woman_facepalming: I realised that I hadn't built the Kernel with BTRFS support. Sorry for making this issue.

closed 2021-02-16 by Ominitay

illiliti, 2021-02-16:
All good :)