illiliti/tinyramfs #25: Root device not found or unavailable (LUKS/LVM)

WanderingFarUnix, 2021-06-20:
I am pretty new to KISS but I wanted to try it out.
I encrypt all my installs due to me being on a laptop
The kernel compiled fine.
 Grub was a different story as the only thing that worked was to grub-install on the live-usb i was using and not the chroot.
Now the issue I am facing is Trying to get LUKS LVM working. 

When the system boots up, it will simply stop at 'Device not found or unavailable' and it will say LUKS decryption has failed. 
I go into recovery mode and it seems to stop at blkid. It seems to be finding the UUID of the root partition but just can't.

I have tried:

- Defining both the root and and crypt root as a UUID and as /dev/by/uuid
- Defining root in GRUB
- Changing mdev to eudev
- Reinstalling GRUB
- Changing the compress to add --check-crc32
- Defining root as a PART-UUID
- Putting the lvm hook before luks

I cannot seem to get anything to work. It does seem to be a problem with detecting the root partition.
The kernel does have Twofish and sha512 embedded into it (as I use that to encrypt my system).
I also checked blkid in the emergency shell that it kicked me into when I try to boot and it shows the encrypted partition.
It shows as /dev/sdb3 with the same UUID i specified in the tinyramfs config (/dev/sda below is /dev/sdb when it boots up which is why I use UUID).

Here is the way I encrypted it (this is shown in the liveusb chrooted)

~~~
NAME              FSTYPE      FSVER            LABEL        UUID                                   FSAVAIL FSUSE% MOUNTPOINT
sda                                                                                                               
├─sda1                                                                                                            
├─sda2            ext4        1.0                           298b044e-2eb9-41d1-b84c-6c6cb78bb226      1.3G     2% /mnt/boot
└─sda3            crypto_LUKS 2                             1c2c7634-12c5-42a7-94b7-c2cf6ba87787                  
  └─kisslvm       LVM2_member LVM2 001                      5A4NCD-n650-RmyN-UC04-2ZM1-y09p-2WTw5o                
    ├─MaoMao-swap swap        1                             f7860e16-871c-4c69-a384-9a961a586ebc                  
    ├─MaoMao-root ext4        1.0                           25efb88a-ba14-4084-a3be-96c28827dd3b    138.5G     0% /mnt
    └─MaoMao-home ext4        1.0                           e7026ea4-dfe7-4c9b-94a5-ebdd864774a1    275.1G     1% /mnt/home

~~~

This is the tinyramfs config:

~~~
compress="xz --check=crc32"
hooks="eudev luks lvm"

luks_discard=1
luks_root=UUID=1c2c7634-12c5-42a7-94b7-c2cf6ba87787

lvm_config=1
lvm_discard=1
lvm_name=root
lvm_group=MaoMao

root=UUID=25efb88a-ba14-4084-a3be-96c28827dd3b
~~~

Grub

~~~
GRUB_DEFAULT=0
GRUB_TIMEOUT=5
GRUB_DISTRIBUTOR="KISS"
GRUB_CMDLINE_LINUX_DEFAULT="loglevel=3 quiet dolvm crypt_root=PARTUUID=bfc72ec0-3e5b-4e0a-8078-724636c4ec14"

# Preload both GPT and MBR modules so that they are not missed
GRUB_PRELOAD_MODULES="part_gpt part_msdos"

# Uncomment to enable booting from LUKS encrypted devices
#GRUB_ENABLE_CRYPTODISK=y

~~~

I am pretty stumped at this issue as it should just boot up fine. Hopefully I can get some help with this.
I am using a KNOPPIX live usb.
I am using coreboot with seabios (All other OS's work so I do not think this could be the issue).
Grub was installed using the live usb onto the kiss installation by mounting /mnt/boot as /boot
The grub config was generated inside the chroot.
Grub boots up fine but the initramfs cannot decrypt the root partition ( as it cannot find one ).

I do suspect it is something to do with the luks hooks as that is the part that fails.

I'll try to get an image of the error if I can.
Thanks!

illiliti, 2021-06-20:
Thanks for very good and detailed report!

Please send me output of `blkid` in emergency shell and working host system.

WanderingFarUnix, 2021-06-21:
Emergency shell:
[image]

Emergency shell with grub recovery mode:
[image]

Working host system via knoppix on live-usb:
[image]

blkid after starting chroot:
[image]

blkid in the chroot:
[image]

List of the binaries in the bin/ folder in the emergency shell:
[image]

Just to note:
- The wronf efi signature is because i am using gpt in BIOS (as I use seaBIOS)
- The mei_me device fails as I used me_cleaner when installing coreboot
- udev seems to throw some warnings. may be related.

commit 93a9cffa (2021-06-24): hooks/luks.init: allow word-splitting to remove empty args

Fixes: #25
closed 2021-06-24 by illiliti