illiliti/tinyramfs #26: Could not mount: No such device (EXT4)

sourTaste000, 2022-03-20:
Hello, I am having problems with tinyramfs. Whenever I boot, it shows this:
```
mount: mounting /dev/disk/by-uuid/* on /mnt/root failed: No such device
!> failed to mount rootfs: /dev/disk/by-uuid/*
sh: can't access tty: job control turned off
```
Picture of error:
[image]

___
[This](https://github.com/archlinux/svntogit-packages/blob/packages/linux-zen/trunk/config) is my kernel config.
This is my tinyramfs config:
```
monolith=0
compress=gzip
root="UUID=14875d4-b8f0-4678-b370-8b873bc67fff"
root_type=ext4
hooks="mdev"
```
My Hardware:
CPU:Intel(R) Core(TM) i5-6200U CPU @ 2.30GHz (Skylake)

GPU:
- VGA compatible controller: Intel Corporation Skylake GT2 [HD Graphics 520] (rev 07)
- 3D controller: NVIDIA Corporation GM108M [GeForce 940M] (rev a2)

RAM: 4GB DDR3

Wireless: 
- Dual Band Wireless-AC 3165 Plus Bluetooth
- Vendor: Intel Corporation

Disk: Model: LITEON CV1-DB256

___
I have looked up these issues:
- https://github.com/illiliti/tinyramfs/issues/22
    - Ext4 is in [external modules](https://github.com/archlinux/svntogit-packages/blob/packages/linux-zen/trunk/config), and the image was built with `monolith=0`, but the issue still appears.
- https://github.com/illiliti/tinyramfs/issues/9
    - Checked hooks, I have `mdev`, but I can't find the ext4 hook. I created the ext4 hook manually, according to the instructions, but when I tried to make the image, it says that the hook is not found even I put it in `hooks.d`.
- https://github.com/illiliti/tinyramfs/issues/25
    - Not using `LUKS` or `LVM`.

___
If you need more information, I will provide them as soon as I can.

illiliti, 2022-03-20:
Run this command in initramfs emergency shell and host system: `blkid`. Then, post the output here.

illiliti, 2022-03-20:
> it says that the hook is not found even I put it in `hooks.d`.

The directory is `hook.d`, not `hooks.d`.
https://github.com/illiliti/tinyramfs/blob/c1c3457fbd57c703226f4f0ece74cb8c339a142e/lib/common.sh#L106

sourTaste000, 2022-03-20:
I can't type in the emergency shell; the built-in keyboard and external keyboard both won't work.

sourTaste000, 2022-03-20:
> > it says that the hook is not found even I put it in `hooks.d`.
> 
> The directory is `hook.d`, not `hooks.d`.
> 
> https://github.com/illiliti/tinyramfs/blob/c1c3457fbd57c703226f4f0ece74cb8c339a142e/lib/common.sh#L106

sorry, I meant `hook.d`.

sourTaste000, 2022-03-20:
I tried to add keymap to the tinyramfs config:
```
monolith=0
compress=gzip
root="UUID=1487e5d4-b8f0-467f-b370-8b873bc67fff"
root_type=ext4
hooks="mdev keymap"
keymap_path=/etc/tinyramfs/us.bmap
```
but it is showing:
```
/etc/tinyramfs # tinyramfs -f -k "5.16.15-zen1-1-zen" /boot/initramfs-5.16.15-zen1-1-zen.img
>> creating ramfs structure
!> unable to find hook: mdev keymap
```

illiliti, 2022-03-20:
The separator of hooks is comma, not space.
```
hooks=mdev,keymap
```

> I can't type in the emergency shell; the built-in keyboard and external keyboard both won't work.

This means modules aren't loaded at all. You need to extract initramfs image and ensure the modules are present there.

sourTaste000, 2022-03-21:
All the modules are missing:
<details>
<summary>Contents inside</summary>
Ignore the `.DS_Store` files.
<pre>
$ find .
.
./proc
./init
./usr
./usr/lib64
./usr/.DS_Store
./usr/bin
./usr/bin/mdev
./usr/bin/sleep
./usr/bin/switch_root
./usr/bin/printf
./usr/bin/env
./usr/bin/kill
./usr/bin/sh
./usr/bin/busybox
./usr/bin/mkdir
./usr/bin/mount
./usr/bin/loadkmap
./usr/bin/find
./usr/bin/ln
./usr/bin/helper
./usr/bin/[
./usr/bin/blkid
./usr/sbin
./usr/lib
./usr/lib/.DS_Store
./usr/lib/tinyramfs
./usr/lib/tinyramfs/common.sh
./usr/lib/tinyramfs/helper.sh
./usr/lib/tinyramfs/init.sh
./usr/lib/tinyramfs/hook.d
./usr/lib/tinyramfs/hook.d/mdev
./usr/lib/tinyramfs/hook.d/mdev/mdev.init.late
./usr/lib/tinyramfs/hook.d/mdev/mdev.init
./usr/lib/tinyramfs/hook.d/keymap
./usr/lib/tinyramfs/hook.d/keymap/keymap.init
./lib64
./.DS_Store
./bin
./sbin
./etc
./etc/mdev.conf
./etc/.DS_Store
./etc/tinyramfs
./etc/tinyramfs/config
./etc/tinyramfs/us.bmap
./var
./var/lock
./var/run
./sys
./mnt
./mnt/.DS_Store
./mnt/root
./root
./lib
./dev
./run
./tmp
</pre>
</details>

sourTaste000, 2022-03-21:
using an non-existent kernel version also works
```
/usr/lib/modules # ls
5.16.15-zen1-1-zen

/usr/lib/modules # tinyramfs -f -k "5.16.15-does-not-exist" /boot/initramfs-5.16.15-zen1-1-zen.img
>> creating ramfs structure
>> copying hook: mdev.init
>> copying hook: mdev.init.late
>> evaluating hook: mdev
>> copying hook: keymap.init
>> evaluating hook: keymap
>> generating initramfs image
+> done: /boot/initramfs-5.16.15-zen1-1-zen.img

/usr/lib/modules # tinyramfs -f -k "5.16.15-zen1-1-zen" /boot/initramfs-5.16.15-zen1-1-zen.img
>> creating ramfs structure
>> copying hook: mdev.init
>> copying hook: mdev.init.late
>> evaluating hook: mdev
>> copying hook: keymap.init
>> evaluating hook: keymap
>> generating initramfs image
+> done: /boot/initramfs-5.16.15-zen1-1-zen.img
```

illiliti, 2022-03-21:
Remove `monolith=0` from config

sourTaste000, 2022-03-21:
It shows that it is copying modules now, but the keyboard still doesn't work. Is it a problem with the kernel config?
```
/ # tinyramfs -f -k "5.16.15-zen1-1-zen" /boot/initramfs-5.16.15-zen1-1-zen.img
>> creating ramfs structure
>> copying hook: mdev.init
>> copying hook: mdev.init.late
>> evaluating hook: mdev
>> copying hook: keymap.init
>> evaluating hook: keymap
>> copying all modules
>> generating initramfs image
+> done: /boot/initramfs-5.16.15-zen1-1-zen.img
```
<details>
<pre>
~/Downloads/u8yv2d
λ find .
.
./proc
./init
./usr
./usr/lib64
./usr/bin
./usr/bin/mdev
./usr/bin/modprobe
./usr/bin/sleep
./usr/bin/switch_root
./usr/bin/printf
./usr/bin/env
./usr/bin/kill
./usr/bin/sh
./usr/bin/busybox
./usr/bin/kmod
./usr/bin/mkdir
./usr/bin/mount
./usr/bin/loadkmap
./usr/bin/find
./usr/bin/ln
./usr/bin/helper
./usr/bin/[
./usr/bin/blkid
./usr/sbin
./usr/lib
./usr/lib/liblzma.so.5
./usr/lib/libz.so.1
./usr/lib/tinyramfs
./usr/lib/tinyramfs/common.sh
./usr/lib/tinyramfs/helper.sh
./usr/lib/tinyramfs/init.sh
./usr/lib/tinyramfs/hook.d
./usr/lib/tinyramfs/hook.d/mdev
./usr/lib/tinyramfs/hook.d/mdev/mdev.init.late
./usr/lib/tinyramfs/hook.d/mdev/mdev.init
./usr/lib/tinyramfs/hook.d/keymap
./usr/lib/tinyramfs/hook.d/keymap/keymap.init
./usr/lib/liblzma.so.5.2.5
./usr/lib/ld-musl-x86_64.so.1
./usr/lib/libz.so.1.2.11
./usr/lib/modules
./usr/lib/modules/5.16.15-zen1-1-zen
./usr/lib/modules/5.16.15-zen1-1-zen/modules.builtin.bin
./usr/lib/modules/5.16.15-zen1-1-zen/modules.builtin.alias.bin
./usr/lib/modules/5.16.15-zen1-1-zen/modules.dep.bin
./usr/lib/modules/5.16.15-zen1-1-zen/modules.alias
./usr/lib/modules/5.16.15-zen1-1-zen/modules.order
./usr/lib/modules/5.16.15-zen1-1-zen/modules.symbols.bin
./usr/lib/modules/5.16.15-zen1-1-zen/modules.dep
./usr/lib/modules/5.16.15-zen1-1-zen/modules.alias.bin
./usr/lib/modules/5.16.15-zen1-1-zen/modules.softdep
./usr/lib/modules/5.16.15-zen1-1-zen/modules.symbols
[... 674 lines cut ...]
./usr/lib/modules/5.16.15-zen1-1-zen/kernel/fs/hfsplus
./usr/lib/modules/5.16.15-zen1-1-zen/kernel/fs/hfsplus/hfsplus.ko.zst
./usr/lib/modules/5.16.15-zen1-1-zen/modules.builtin
./usr/lib/libc.so
./lib64
./bin
./sbin
./etc
./etc/mdev.conf
./etc/tinyramfs
./etc/tinyramfs/config
./etc/tinyramfs/us.bmap
./var
./var/lock
./var/run
./sys
./mnt
./mnt/root
./root
./lib
./dev
./run
./tmp
</pre>
</details>

illiliti, 2022-03-21:
I've pushed fix that adds keyboard modules to initramfs image. Please try latest master.

sourTaste000, 2022-03-21:
they are all there, but the keyboard is still not working :(
```
drivers/input/keyboard
λ ls
adc-keys.ko.zst			iqs62x-keys.ko.zst		qt1050.ko.zst
adp5520-keys.ko.zst		lkkbd.ko.zst			qt1070.ko.zst
adp5588-keys.ko.zst		lm8323.ko.zst			qt2160.ko.zst
adp5589-keys.ko.zst		lm8333.ko.zst			samsung-keypad.ko.zst
applespi.ko.zst			matrix_keypad.ko.zst		stowaway.ko.zst
atkbd.ko.zst			max7359_keypad.ko.zst		sunkbd.ko.zst
cros_ec_keyb.ko.zst		mcs_touchkey.ko.zst		tca6416-keypad.ko.zst
cypress-sf.ko.zst		mpr121_touchkey.ko.zst		tca8418_keypad.ko.zst
dlink-dir685-touchkeys.ko.zst	mtk-pmic-keys.ko.zst		tm2-touchkey.ko.zst
gpio_keys.ko.zst		newtonkbd.ko.zst		twl4030_keypad.ko.zst
gpio_keys_polled.ko.zst		opencores-kbd.ko.zst		xtkbd.ko.zst
```

sourTaste000, 2022-03-21:
The usb port is not working either; When I plug in the external keyboard, the lights(caps lock, scroll lock, etc) do not show up.

illiliti, 2022-03-21:
Ah, i think i know why modules aren't loading. busybox modprobe doesn't support zstd-compressed modules. You need to either swap modprobe to kmod(not sure if it's support zstd though) or use xz/gzip for compression.

sourTaste000, 2022-03-21:
Looks like kmod already has support for zstd since version 28. The current version of kmod that is installed on my laptop is version 29. I switched all busybox module commands to kmod's commands, and I did `depmod 5.16.15-arch1-1`. It did not give out any warning or error messages, so I think it worked?

By the way, is there a way to copy _every_ module to the tinyramfs image?

illiliti, 2022-03-21:
> By the way, is there a way to copy _every_ module to the tinyramfs image?

The size of initramfs would be too big if tinyramfs copied every module. By default, only essential modules are copied. However, you can write hook that copies every module.

sourTaste000, 2022-03-21:
Sorry for bothering you with this issue :p I'll stop being lazy and built the kernel myself with a different compression format 😅
(The kernel and modules that I am using is downloaded from arch linux's mirror if you haven't noticed)

closed 2022-03-21 by sourTaste000

sourTaste000, 2022-03-26:
Issue is now resolved! Looks like kmod was not built with zstd support. I have submitted an issue and hope the maintainer of that package can fix it.

sourTaste000, 2022-03-26:
[image]