illiliti/tinyramfs #32: automatic detection of root device

wael444, 2022-07-16:
would this be possible?

illiliti, 2022-07-17:
It is possible, but that would be unreliable.

root device autodetection implies that we have to iterate over disk devices, mount them and then check /sbin/init. if it exist - switch_root into it, if not - continue.

It is unreliable because mount can fail due to fs module not being loaded. mount can also fail because disk device could be encrypted, so have to implement fs type checking. There're also more complex file systems like zfs or btrfs, which may have nested file systems/partitions/pools...

This is pretty hard to do properly. I'd rather have only static configuration defined by user. This is more reliable than being too smart and autodetect everything.

> since root device can be automatically detected, so can the file system module for it.

Tinyramfs already autodetects filesystem module for rootfs: https://github.com/illiliti/tinyramfs/blob/6b9f73dbc0af7ea4e2e5a8944a92368c91c1561c/tinyramfs#L112-L120

But again, it is unreliable. It might be removed in the future because it doesn't work in live usb environment and environment where /proc isn't mounted.

wael444, 2022-11-29:
not exactly what i meant, what i meant was that at configuring time when tinyramfs is building the initramfs. i wouldn't think this is that hard because it can simply check what device is `/`.