illiliti/tinyramfs #34: could not open modules.builtin.modinfo & cannot unlock LUKS

dkwo, 2023-03-28:
on Void Linux, using https://github.com/void-linux/void-packages/pull/43061 and
```
$ lsblk 
NAME                 MAJ:MIN RM   SIZE RO TYPE  MOUNTPOINTS
nvme0n1              259:0    0 476.9G  0 disk  
├─nvme0n1p1          259:1    0   512M  0 part  /boot
└─nvme0n1p2          259:2    0 476.4G  0 part  
  └─cryptroot        254:0    0 476.4G  0 crypt 
    ├─cryptroot-root 254:1    0    15G  0 lvm   /
    └─cryptroot-home 254:2    0 461.4G  0 lvm   /home
```
the config (dots are replaced by actual value of `blkid /dev/nvme0n1p2`)
```
$ cat /etc/tinyramfs/config 
compress="gzip -9"
hooks="eudev,luks,lvm"
luks_root="UUID=..."
luks_name=cryptroot 
luks_discard=yes
lvm_group=cryptroot 
lvm_name=root$
```
produces
```
$ doas xbps-reconfigure -f linux6.1 
linux6.1: configuring ...
Executing post-install kernel hook: 20-initramfs ...
>> creating ramfs structure
>> copying hook: eudev.init
>> copying hook: eudev.init.late
>> evaluating hook: eudev
>> copying hook: luks.init
>> evaluating hook: luks
>> copying hook: lvm.init
>> evaluating hook: lvm
>> copying all modules
depmod: WARNING: could not open modules.builtin.modinfo at /tmp/tinyramfs.11858/lib/modules/6.1.21_1: No such file or directory
>> generating initramfs image
+> done: boot/initramfs-6.1.21_1.img
Executing post-install kernel hook: 50-efibootmgr ...
linux6.1: configured successfully.
```
and then the boot is stuck at
```
Enter passphrase for /dev/disk/by-uuid/...:
device-mapper: table: 254:0 crypt: unknown target type
device-mapper: ioctl: error adding target to table
device-mapper: reload ioctl on cryptroot (254:0) failed: Invalid argument
!> failed to unlock LUKS
```
Is this related to the patch mentioned for kmod? or something else?
Thanks.

illiliti, 2023-03-28:
First of all thank you for bringing tinyramfs to voidlinux!

`unknown target type` usually means that some module isn't loaded or missing. in most cases it's `dm*` modules but technically can be anything. Not sure how this relates to modinfo file because things were OK without it... Hmm yes, can you try this patch?

```diff
diff --git a/tinyramfs b/tinyramfs
index f2f1d96..7b7a6b0 100755
--- a/tinyramfs
+++ b/tinyramfs
@@ -143,6 +143,9 @@ copy_modules()
     copy_file "${moddir}/${kernel}/modules.builtin" \
               "/lib/modules/${kernel}/modules.builtin" 0644
 
+    copy_file "${moddir}/${kernel}/modules.builtin.modinfo" \
+              "/lib/modules/${kernel}/modules.builtin.modinfo" 0644
+
     depmod -b "$tmpdir" "$kernel"
 }
 
```

dkwo, 2023-03-29:
thanks, the patch fixes the warning, but not the main issue.
the initram contains `/usr/lib/modules/6.1.21_1/kernel/drivers/md/dm-crypt.ko.zst`.
do you see anything wrong with my config or the way tinyramfs is packaged?

illiliti, 2023-03-29:
No, the package and config looks good to me. I think the best way to figure out what's wrong is to boot with debug mode(`rddebug` kernel command line parameter) enabled and check which modules are loaded(maybe they are not at all) using `kmod list` or /proc/modules.

dkwo, 2023-03-29:
This is what I get:
```
sh: 0: can't access tty: job control turned off
# kmod list
Module Size Used by
dm_mod 155648 0
crct10dif_pclmul 16384 1
crc32_pclmul 16384 0
crc32c_intel 24576 0
polyval_clmulni 16384 0
polyval_generic 16384 1 polyval_clmulni
gf128mul 16384 1 polyval_generic
ghash_clmulni_intel 16384 0
sha512_ssse3 49152 0
aesni_intel 393216 0
crypto_simd 16384 1 aesni_intel
cryptd 24576 2 crypto_symd,ghash_clmulni_intel
rapl 16384 0
intel_cstate 20480 0
```

illiliti, 2023-03-30:
What if you manually load dm-crypt module with modprobe?

dkwo, 2023-03-30:
```
modprobe dm_crypt
modprobe: ERROR: could not insert 'dm_crypt': Unknown symbol in module, or unknown parameter (see dmesg)
dmesg
sh: 2: dmesg: not found
```

illiliti, 2023-03-31:
Put `copy_exec dmesg` into /etc/tinyramfs/hook.d/dmesg, regenerate initramfs and try again

dkwo, 2023-03-31:
that did not work, I had to edit by hand /usr/bin/tinyramfs
then dmesg gets a ton of unknown symbols, a video is worth a thousand words https://paste.c-net.org/WorkingsIsraeli

illiliti, 2023-04-01:
> that did not work, I had to edit by hand /usr/bin/tinyramfs

My bad! I gave you wrong path. It should be `/etc/tinyramfs/hook.d/dmesg/dmesg`. You also need to append dmesg to hooks in config

> then dmesg gets a ton of unknown symbols, a video is worth a thousand words

Either some module is missing or has wrong version(unlikely). You will have to find out why.

For example you can use `modprobe -D` to locate module dependencies and check if some are missing.

dkwo, 2023-04-01:
Thanks. At least `xhci_hcd` and `dm_crypt` have unknown symbol errors from the video.
I do not see any `-D` option in modprobe's man page in Void. do you mean `--show-depends` ?

dkwo, 2023-04-01:
sorry, ignore my comment about -D.

dkwo, 2023-04-01:
on the working system:
```
$ modprobe -D dm_crypt 
insmod /lib/modules/6.2.8_1/kernel/drivers/char/hw_random/rng-core.ko.zst 
insmod /lib/modules/6.2.8_1/kernel/drivers/char/tpm/tpm.ko.zst 
insmod /lib/modules/6.2.8_1/kernel/drivers/tee/tee.ko.zst 
insmod /lib/modules/6.2.8_1/kernel/lib/asn1_encoder.ko.zst 
insmod /lib/modules/6.2.8_1/kernel/security/keys/trusted-keys/trusted.ko.zst 
insmod /lib/modules/6.2.8_1/kernel/security/keys/encrypted-keys/encrypted-keys.ko.zst 
insmod /lib/modules/6.2.8_1/kernel/drivers/md/dm-mod.ko.zst 
insmod /lib/modules/6.2.8_1/kernel/drivers/md/dm-crypt.ko.zst 
```
while from the initram shell
```
$ modprobe -D dm_crypt 
insmod /lib/modules/6.1.21_1/kernel/drivers/md/dm-mod.ko.zst 
insmod /lib/modules/6.1.21_1/kernel/drivers/md/dm-crypt.ko.zst
```
so it appears some dependencies are missing.

dkwo, 2023-04-01:
similarly
```
$ modprobe -D xhci_hcd
insmod /lib/modules/6.2.8_1/kernel/drivers/usb/common/usb-common.ko.zst 
insmod /lib/modules/6.2.8_1/kernel/drivers/usb/core/usbcore.ko.zst 
insmod /lib/modules/6.2.8_1/kernel/drivers/usb/host/xhci-hcd.ko.zst 
```
vs
```
$ modprobe -D xhci_hcd
insmod /lib/modules/6.1.21_1/kernel/drivers/usb/host/xhci-hcd.ko.zst 
```

dkwo, 2023-04-01:
Something is wrong with the way it searches for modules.
From the initram:
```
modprobe rng_core
modprobe: FATAL: Module rng_core not found in directory /lib/modules/6.1.21_1
```

dkwo, 2023-04-01:
perhaps adding the folders
```
drivers/char/hw_random
drivers/char/tpm
drivers/tee
lib
security/keys/trusted-keys
security/keys/encrypted-keys
drivers/usb/common
drivers/usb/core
```
at line 128?

illiliti, 2023-04-01:
Try this
```diff
diff --git a/tinyramfs b/tinyramfs
index f2f1d96..854ea40 100755
--- a/tinyramfs
+++ b/tinyramfs
@@ -130,7 +130,8 @@ copy_modules()
                 drivers/usb/storage drivers/input/keyboard -type f 2> /dev/null |
 
             while read -r _mod; do
-                copy_file "$_mod" "/lib/modules/${kernel}/kernel/${_mod}" 0644
+                _mod=${_mod##/*}
+                copy_kmod "${_mod%%.*}"
             done
         )
     fi
```

dkwo, 2023-04-01:
Thanks, let me try and report.
Let me add that `hostonly` also does the job (somehow I also missed root=...)
```
$ cat /etc/tinyramfs/config 
compress="gzip -9"
hostonly=true
hooks="eudev,luks,lvm"
luks_root="UUID=..."
luks_name=cryptroot 
luks_discard=yes
lvm_group=cryptroot 
lvm_name=root
root="UUID=..."
```

dkwo, 2023-04-01:
with the patch (and without hostonly), i still get:
```
Cannot initialize device-mapper. Is dm_mod kernel module loaded?
Cannot use device cryptroot, name is invalid or still in use.
```

illiliti, 2023-04-01:
Is there anything bad in dmesg?

dkwo, 2023-04-01:
dmesg goes so fast I cannot see any error, it even goes on to starting init and udev. how should i parse it?
however, it seems something is still wrong with folders, as
```
modprobe dm_crypt
modprobe: FATAL: Module dm_crypt not found in directory /lib/modules/6.1.21_1
```

illiliti, 2023-04-01:
Same error will popup if you run modprobe with `-D` flag, right? What if you do modprobe dm_mod?

dkwo, 2023-04-01:
yes, same error for both.

dkwo, 2023-04-07:
if i add the line `drivers/char drivers/tee security/keys \` to find, then my system boots fine also without hostonly.
do you think this can be patched into tinyramfs, as well as the patch for modules.builtin.modinfo?
btw, this is now merged in void linux :)

illiliti, 2023-04-15:
I need to reconsider and perhaps redo how tinyramfs copies modules. I'll look into it when i get more time.

dkwo, 2023-04-15:
that makes sense, thank you. another option is to add those drivers to the luks hook. i'm enjoying tinyramfs btw :)

closed 2024-09-22 by illiliti