illiliti/tinyramfs #36: init.late hook has issue with cp -r /folder/*

dkwo, 2023-04-14:
as part of https://github.com/void-linux/void-packages/pull/39796 i have a custom late init hook that loads firmware into a tmpfs.
the hook works fine in its current form
```
[ -e /vendorfw ] || return 0
print 'asahi: copying vendor firmware to tmpfs under root filesystem'
loaderdir=/mnt/root/lib/firmware/vendor
mkdir -p "$loaderdir"
mount -t tmpfs vendorfw "$loaderdir"

cp /vendorfw/manifest.txt "$loaderdir"
cp -r /vendorfw/asmedia "$loaderdir"
cp -r /vendorfw/brcm "$loaderdir"
```
but if I replace the three `cp` invocations with `cp -r /vendorfw/* "$loaderdir"`, then the boot process hangs indefinitely.
do you know what the issue could be?

illiliti, 2023-04-15:
https://github.com/illiliti/tinyramfs/blob/b625eee02f696303c2ee9f5266421d16103e3831/lib/init.sh#L99

Globbing is disabled by default. You need to temporarily reenable it to use `*`

dkwo, 2023-04-15:
i see, many thanks.

closed 2023-04-15 by dkwo