illiliti/tinyramfs #37: cp: cannot stat 'module option': No such file or directory

dkwo, 2023-05-28:
With the following
```
$ cat /etc/modprobe.d/modprobe.conf 
options snd-hda-intel model=tpt460
```
I do get an error, and the image is not created:
```
$ doas xbps-reconfigure -f linux6.3
linux6.3: configuring ...
Executing post-install kernel hook: 20-initramfs ...
>> creating ramfs structure
>> copying hook: eudev.init
>> copying hook: eudev.init.late
>> evaluating hook: eudev
>> copying hook: luks.init
>> evaluating hook: luks
>> copying hook: lvm.init
>> evaluating hook: lvm
>> copying hostonly modules
cp: cannot stat '/lib/modules/6.3.3_1/kernel/sound/pci/hda/snd-hda-intel.ko.zst model=tpt460': No such file or directory
Executing post-install kernel hook: 50-efibootmgr ...
linux6.3: configured successfully.
```
It seems that tinyramfs is incorrectly parsing my modprobe.d configuration.

illiliti, 2023-05-28:
Does this patch fix the bug for you?

```diff
diff --git a/lib/common.sh b/lib/common.sh
index 1f72a69..db7477a 100755
--- a/lib/common.sh
+++ b/lib/common.sh
@@ -95,7 +95,7 @@ copy_kmod()
 {
     modprobe -S "$kernel" -D "$1" 2> /dev/null |
 
-    while read -r _ _mod || [ "$_mod" ]; do
+    while read -r _ _mod _ || [ "$_mod" ]; do
         case $_mod in /*) copy_file "$_mod" "$_mod" 0644; esac
     done
 }

```

dkwo, 2023-05-28:
it works, thank you!
now the initram is generated, and audio works fine.
can you merge this into master, so i can add the patch to void linux repos?

closed 2023-05-28 by dkwo

illiliti, 2023-05-28:
https://github.com/illiliti/tinyramfs/commit/80074e02dde40fc25af66098c45984f0e4ea5e3c