illiliti/tinyramfs #39: Generate initramfs for live ISO

firasuke, 2023-06-30:
Hey there,

I am trying to use `tinyramfs` to generate an initramfs for a custom distribution I am working on that uses `musl`, `toybox`, `mdevd` and `libudev-zero`.

Does `tinyramfs` support hardware detection at boot (not at initramfs generation time), as this will be a distribution initramfs that should be able to detect and load accordingly?

Are their any pointers or guides to get this working as intended?

Thanks in advance!

illiliti, 2023-07-04:
Hi! tinyramfs does support hardware detection at boot as long as you include all modules necessary for booting. As of how to make this all work, there is the [kiss-live](https://github.com/kiss-community/kiss-live)  project which uses tinyramfs to generate initramfs for live iso. You can look into how it works and make something similar for youself.

firasuke, 2023-07-04:
Interesting, thanks for your input on this.

Keep up the great work!

closed 2023-07-04 by firasuke