illiliti/tinyramfs #45: Can't mount root...

7z3m, 2024-08-22:
I'm getting a no such file or directory error. However I can mount manually on the command line with no errors, and blkid shows every single device in /dev accurately. 

I have devtmpfs builtin the kernel, I'm not using udev/mdev. I have no encryption applied and everything builtin to the kernel, with the exeption of wifi drivers.

I checked every issue similar to this, to no resolve.

this is my config file:

*root="UUID=..."
root_type="ext4"
hostonly="yes"
compress="cat"
monolith="yes"

*replace uuid with actual number.

illiliti, 2024-08-22:
`UUID=...` requires device manager to be present. You can specify `/dev/sda<n>` or `/dev/nvme<n>` if you don't have device manager.

7z3m, 2024-08-23:
ok thanks, i replaced uuid with /dev/sd* and it worked.
sorry for the inconvience.

illiliti, 2024-08-23:
No worries

closed 2024-08-23 by illiliti